I-Connect/NukiBleEsp32 #7: First feedback

peteh, 2022-04-18:
Hi, 

I was the one responding to your forum entry in nuki forums for trying out your lib. 

I'm actually very impressed. I managed to use it to create a mqtt lock integration with homeassistant as an afternoon exercise. (https://github.com/peteh/doorauth)

A few things that I noticed when using the lib: 
 - there is no documentation of the method signatures, probably due to the not yet polished state of the lib - I think all public methods should have an explanation of each parameter (they match mostly the signatures of the nuki dev document, but having it at hand in auto complete would be much easier)
 - method names are not always self explanatory, e.g. setSecurityPin does actually set a new pin on the lock while saveSecurityPin saves the pin for the communication - here the aforementioned documentation would have helped
 -  the security pin seems to be mandatory for the lib while not all actions actually need it, technically a bridge that just opens/closes the lock does not need the pin in my understanding of the concept. However, the lib will fail with INVALID_PIN error on connection

In general I still have a few questions, even though my first tries pretty much worked out of the box: 

How is the bluetooth connection supposed to work? We have a SmartLockEventHandler that gets updated when the lockstate changes, however, how is it getting notified? Does this happen during the bluetooth scanning? 

How often is the bluetooth scanning necessary? The demo app does it pretty much cyclically without waiting much, but as I understand this would cause pretty strong battery strain on the lock itself.  I'm unfortunately not that deep into bluetooth yet, what is the purpose of the scanning and doing it very often to begin with? I have seen the same with RaspiNukiBridge but esp32 seems to be much more stable. 

Again, thank you very much for this huge amount of work. A few month back I thought about starting a similar project but realized pretty quickly that this would require many hours to get right.

I-Connect, 2022-04-19:
Hi Pete,

Thx for the compliment, good to hear you could easily get it working.

Documenting the methods is indeed on the list.

I will think about the mandatory security pin, it is needed for setting config changes/adding users etc...

Connecting and scanning are two different processes.
- The ESP establishes a new BT connection every time a command is sent, when no data is sent anymore the lock times out the connection.
- Scanning goes on continuously on the ESP with intervals chosen in such a way that it will never miss an advertisement sent from the lock.
- The lock always continuously sends advertisements (the interval is a setting in the config ( CmdResult setAdvertisingMode(AdvertisingMode mode); ), this interval determines the battery drain on the lock). When the lock state is changed a parameter is changed in the advertisement. This causes the SmartLockEventHandler to be triggered and then you could initiate a follow up like requesting the keyturner state.
- (I will document this in the Readme...)

Most welcome, it was indeed a huge amount of work but with nice results and learned a lot again on encryption and BT :-)

I-Connect, 2022-04-19:
Depends what you want to query. If I understand correctly the lock advertises changes in the keyturner state but ie not the config changes.

So if you are only interested in the keyturner state changing waiting for the SmartLockEventHandler  is enough.
(and will be more battery efficient as you do not need to create a BT connection)

For your second question, this was answered by Nuki:
https://developer.nuki.io/t/bluetooth-specification-questions/1109/26

> They are between ~400ms and ~1s.

peteh, 2022-04-23:
Thanks for clarification. 

As i understand the default settings in the sample app are optimal then and i should not need to force a state read. 

However, I'm not sure if the lock state gets updated in the sample app directly after executing a lock action. That's why i added the forced state read after executing a lock action from my app. But I'll have another try without it. And experiment. So far it works much better than raspinukibridge that i used before. 

Technically the issue could be closed except if you want to use it for tracking of the api documentation or something else.

I-Connect, 2022-04-23:
oke, will close it

closed 2022-04-23 by I-Connect