I-Connect/NukiBleEsp32 #8: Incorrect battery critical and charging indications

uriyacovy, 2022-04-19:
Hi,
First of all, very impressive module! I am developing an ESPHome component based on your work, and hopefully would be able to post it soon.

I've noticed that it was not clear whether the battery critical and battery charging indications are LSB or MSB. However, considering the battery level (%) value, it makes more sense that it would be LSB. Meaning the first bit is "Smart Lock Battery State Critical" and the 2nd is "Charging Flag".
Let me know if you want me to open a PR.

peteh, 2022-04-19:
I noticed this too, that the flag was 1 but soc was 84% and the RaspiNukiBridge was not reporting critical battery, so I can confirm this issue.

I-Connect, 2022-04-19:
@uriyacovy  thx for the compliment. Feel free to open a PR

Jeroen

uriyacovy, 2022-04-21:
Done - https://github.com/I-Connect/NukiBleEsp32/pull/9

closed 2022-04-23 by I-Connect