I-Connect/NukiBleEsp32 #11: Seems not to compile anymore in platformio with espressif32 > 4.0

peteh, 2022-04-25:
4 days ago version 4 was released for espressif32 framework. I updated and now can't compile with the lib anymore with the following error messages: 

```
> Executing task: platformio run --environment lolin_d32 <

Processing lolin_d32 (platform: espressif32; board: lolin_d32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_d32.html
PLATFORM: Espressif 32 (4.1.0) > WEMOS LOLIN D32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.20001.0 (2.0.1) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-openocd-esp32 2.1100.20220411 (11.0) 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <ArduinoJson> 6.19.4
|-- <Nuki Ble for Esp32> 0.0.4
|   |-- <NimBLE-Arduino> 1.3.6+sha.fc746b1
|   |-- <CRC16> 0.0.0+20220417211510.sha.f46d86b
|   |-- <Preferences> 2.0.0
|-- <ArduinoOTA> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ESPmDNS> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <mqttdevice>
|   |-- <ArduinoJson> 6.19.4
|-- <WiFi> 2.0.0
Building in release mode
Compiling .pio/build/lolin_d32/src/main.cpp.o
Compiling .pio/build/lolin_d32/libea9/PubSubClient/PubSubClient.cpp.o
In file included from src/main.cpp:7:
.pio/libdeps/lolin_d32/Nuki Ble for Esp32/src/NukiBle.h:9:10: fatal error: FreeRTOS.h: No such file or directory

******************************************************************
* Looking for FreeRTOS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:FreeRTOS.h"
* Web  > https://registry.platformio.org/search?q=header:FreeRTOS.h
*
******************************************************************

 #include "FreeRTOS.h"
          ^~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLE2904.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEAddress.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEAdvertisedDevice.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEAdvertising.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEBeacon.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLECharacteristic.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEClient.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEDescriptor.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEDevice.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEEddystoneTLM.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEEddystoneURL.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEExtAdvertising.cpp.o
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLEHIDDevice.cpp.o
*** [.pio/build/lolin_d32/src/main.cpp.o] Error 1
Compiling .pio/build/lolin_d32/lib3c5/NimBLE-Arduino/NimBLERemoteCharacteristic.cpp.o
========================================================================== [FAILED] Took 3.05 seconds ==========================================================================
The terminal process "platformio 'run', '--environment', 'lolin_d32'" terminated with exit code: 1.
```

In the changelog I don't see much that could pinpoint to the issue but it seems like the freertos includes are not available anymore.

Bascy, 2022-04-26:
I've set the platform version explicitly in platform.ini and corrected for the breaking changes that this version has
Can you test again?

peteh, 2022-04-26:
Do you know how I can force platformio to use the master instead of the 4.0.0 tag? 
I'm including the lib via platformio.ini

Bascy, 2022-04-26:
You can specifiy platofrm version in platformio.ini like this:

`platform = espressif32@^4.0.0`

This will use the latest version smaller than 5.0.0

peteh, 2022-04-26:
I meant for your lib, sorry. It uses the platform automatically after an update, but your lib, pulled through platformio is not what you commited to the master but the latest 0.0.4 (I messed up the versions)

I currently have: 
	bascy/Nuki Ble for Esp32@^0.0.4 in
in my platformio.ini

Bascy, 2022-04-26:
No you have to add the platform version to the `platform` property in [env] section of the platformio.ini

peteh, 2022-04-26:
I think we are talking two different things. 
I am not building your lib, I am using my own project which pulls your lib through platformio via dependency. 
i.e. this: 
https://github.com/peteh/doorauth/blob/main/doorauth/platformio.ini

If you build my project with platformio and espressif32 v4.x it will fail because version 0.0.4 version of NukiBleEsp32 lib will look for FreeRTOS.h still (as it is only available in 0.0.4 version) :+1: 
[image]

However, if I build nukible with the current master, it works fine. So the changes on your master/main seem to be at least buildable. But I don't know how I can include the new version of your code in my own project because I usually just add the lib deps in platformio and it pulls the latest official release.

Bascy, 2022-04-26:
Ah, ok, I get you now
You can specify a dependency also by its github url, so if you add `https://github.com/I-Connect/NukiBleEsp32` to your dependencies it will take the latest version from main branch

peteh, 2022-04-26:
Thanks, that works. I built it and deployed it with my project. So I think this issue can be closed. But there probably needs to be a 0.0.5 published for platformio. 

So far it seems to work. I can lock and unlock the door with the latest master.

closed 2022-04-29 by peteh