I-Connect/NukiBleEsp32 #15: Support for multiple Nuki locks

Savjee, 2022-07-21:
Hi,

First up: amazing work on this library! It works very well and means that I don't need to buy expensive Nuki bridges.

I'm attempting to add support for multiple Nuki locks on a single ESP32, using [this ESPHome implementation](https://github.com/uriyacovy/ESPHome_nuki_lock/) which is based on your library.

I'm thinking about creating multiple instances of `Nuki::NukiLock`, but I believe that won't work because your library is using fixed names for storing the preference (eg: `BLE_ADDRESS_STORE_NAME`).

I'll try to namespace this key, but I wanted to check if you have additional thoughts on this.
I have zero experience with BLE, so any thoughts or comments would be appreciated!

I-Connect, 2022-09-06:
Hi Savjee, sorry for the delayed response, holidays and a bit busy...

We should not spawn different BLE scanners, the idea is that 1 BLE scanner handles all traffic (also from other BLE sensors).

@technyon Do you maybe have any thoughts on how to fix storing preferences for multiple locks? Might be a good addition.

technyon, 2022-09-06:
Hi,

Have a look at the constructor of NukiBle:

```
NukiBle::NukiBle(const std::string& deviceName,
                 const uint32_t deviceId,
                 const NimBLEUUID pairingServiceUUID,
                 const NimBLEUUID deviceServiceUUID,
                 const NimBLEUUID gdioUUID,
                 const NimBLEUUID userDataUUID,
                 const std::string preferencedId)
```

This id is later used to initialize preferences:

`preferences.begin(preferencesId.c_str(), false);`

NukiBle is the base class and is inherited by NukiLock (and NukiOpener). The constructor for NukiLock takes a device id which is then fed to the base class as the preferences Id. As long as you use unique device names, it should be no problem to have multiple locks. Just make sure to register every instance of NukiLock for the BleScanner (call NukiLock::registerBleScanner() ).

workhard101, 2022-09-26:
Hi,

I have a question about the library , and I flashed the project  on my "ESP32-S3-Devkitc-1" Board. 

I think now the "ESP32-S3" Board acts as a NUKI Accessories(for example NUKI FOB).

How can I chang the library to let it(ESP32-S3) be acted as ein NUKI Lock, so that I can use a real NUKI FOB to controll the NUKI Lock(ESP32-S3)?

technyon, 2022-09-26:
This isn't what this library does. It is a client to the lock, so you have to pair the fob with the lock.

workhard101, 2022-09-26:
thank you for the feedback! I have a litte confuse about it. 

I have a  "ESP32-S3-Devkitc-1" and a real NUKI FOB. My destination is that: I can controll the ESP32 Boad with a real NUKI FOB, same like a real NUKI FOB can controll a real NUKI Lock.

 As I said, I have flashed the project on  "ESP32-S3-Devkitc-1", and I have tried to pair the "ESP32-S3-Devkitc-1" with a real  NUKI FOB. after pressing 5 seconds on button of "NUKI FOB", but I always got as follow logging. no pair with each other,
How should I change the project, to let NUKI FOB be paired with "ESP32-S3-Devkitc-1"?

D NimBLEScan: >> start: duration=3
D NimBLEScan: << start()
[  3324][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  3325][D][NukiBle.cpp:102] pairNuki(): No nuki in pairing mode found
[  3332][D][NukiBle.cpp:107] pairNuki(): pairing result 0
I NimBLEScan: New advertiser: 41:96:40:5e:45:cb
I NimBLEScan: Updated advertiser: 41:96:40:5e:45:cb
I NimBLEAdvertisedDevice: No service data found
I NimBLEScan: New advertiser: 1e:2f:85:3e:7b:65
I NimBLEScan: New advertiser: de:d2:a5:c3:ba:0b
I NimBLEScan: New advertiser: 6e:5d:ff:7b:d9:0a
...

technyon, 2022-09-26:
The library controls the lock. When you call pairNuki(), it's looking for a lock in pairing mode. Since your Fob isn't a lock, it can't be paired.

Connecting to a fob and receiving button presses from a fob is not the scope of this library.

workhard101, 2022-09-26:
thanks again for your explain! It's clear for me now!

Is It possible/realistic to change something on the base of this library to realize my destination: pairing and controlling the ESP32 Boad with a NUKI FOB?

I am reading another project https://github.com/as19git67/simnuki about "Smart Lock Simulator", but it is in java..., 

(Maybe) Do you know, how should I chang or add some thing to this library to realize my destination?

technyon, 2022-09-26:
I think it's a better idea to create a new library, or put everything into a seperate class. You have to implement the opposite side of the equasion, the server and not the client, including all the authentication, encryption and so on. Check the official API documentation, but it's nothing that's can be just changed in this library, you'll have to write most of it from scratch.

What is you reason that you want to pair the fob with the ESP32? You can just pair it with the Lock and it does what it's supposed to do.

workhard101, 2022-09-26:
thank you for your suggestion/idea!

I have used ESP32-S3 as Bluetooth communication in a project, and  I need a bluetooth keychain FOB to controll it.
I choose NUKI FOB to realize it.

peteh, 2022-09-26:
@workhard101 Do you actually need the nuki lock at all? It seems like you are trying to use the fob for some other project. Depending on your requirements it might not be the right tool. Implementing the counter part for the fob might be a lot of effort and not something that somebody else has already done. You might be better off with something more simple (e.g. 433mhz rf remotes) if security is not a requirement but just to get button presses into an esp32. 

What is the project you are trying to accomplish? 

PS: you're somewhat highjacking this ticket which is not about fob support. So maybe create a new one or we should move the discussion somewhere else.

I-Connect, 2022-09-26:
I agree with peteh, I think a 433 remote is your way forward and ver, very, very less complicated than trying to communicate with a Nuki Fob. Maybe there are even simple ble fobs without encryption which would be handy as the esp already has ble onboard.

As technyon stated a Nuki fob is meant to be paired with a Nuki lock, I doubt it is even doable as there is no api documentation for connecting/communicating directly to a Fob.

Let me know if you have any question otherwise I will close the issue as you are indeed hijacking it a bit :-)

workhard101, 2022-09-27:
@peteh @I-Connect thank you very much for your suggestion!
I need the encryption/security as requirment and must also use the NUKI FOB as requirment 

@I-Connect  as you said "As technyon stated a Nuki fob is meant to be paired with a Nuki lock, I doubt it is even doable as there is no api documentation for connecting/communicating directly to a Fob."

I am reading another project https://github.com/as19git67/simnuki about "Smart Lock Simulator", if I do the NUKI Lock on ESP32-S3 Borad and get button presses from NUKI FOB into an ESP32-S3 Board. In this case, is NUKI FOB API necessary? 

@peteh @I-Connect sorry for hijacking this ticket, we can move the discussion somewhere else.

peteh, 2022-09-27:
@workhard101 please describe the project and what it should do and why you want to achieve, otherwise we can't help you.

technyon, 2022-09-27:
> @workhard101 please describe the project and what it should do and why you want to achieve, otherwise we can't help you.

And open a seperate issue in the process please.

closed 2022-09-27 by I-Connect