I-Connect/NukiBleEsp32 #22: communication between ESP32-S3-Devkitc-1 and NUKI FOB

workhard101, 2022-09-27:
(just continue/move the issue from https://github.com/I-Connect/NukiBleEsp32/issues/15#issuecomment-1259032195)

I want to build a communication between ESP32-S3-Devkitc-1 and NUKI FOB. if I press the buttom of NUKI FOB, a GPIO of ESP32 Board can be set as HIGH or Low.

Using NUKI FOB and encryption/security is requirement. My question:
1. I have also seen that there is no NUKI FOB API Dokument, is it doable for this project?
2. If I do a Simulate on ESP32, let ESP32 act as NUKI Lock, similar to https://github.com/as19git67/simnuki. is it realistic/right way for my project? Is NUKI FOB API Dokument necessary in this case?

peteh, 2022-09-27:
What problem do you want to solve? Or in other words, what is your use case? 
What stops you from using any other remote other than nuki fob if you don't use it with a nuki lock?

workhard101, 2022-09-27:
It's the requirment of company, and we want to buy NUKI FOB and use NUKI FOB directly for our big projeckt. what I am doing now is small part of total projeckt.

The ESP32 as Bluetooth communication is already used in total project, and NUKI FOB is be chosen by company, to act as bluetooth remote controller.

technyon, 2022-09-27:
You'll have to implement a state machine for pairing first I think. In the Simnuki porject this is pairing-general-data-io-characteristic.js, starting from line 93 (PairingGeneralDataInputOutputCharacteristic.prototype.onWriteRequest). Before starting the state machine, you have to announce a lock in pairing mode as an iBeacon.

workhard10, 2022-09-27:
@technyon thank you for your suggestion, and  I am trying..

workhard101, 2022-09-27:
@technyon 
Hi, can you explain it with more detail? about follow what you said:

> "starting from line 93 (PairingGeneralDataInputOutputCharacteristic.prototype.onWriteRequest). Before starting the state machine, you have to announce a lock in pairing mode as an iBeacon"

I use two ESP32-S3-Devkitc-1 Boards, one acts as Server, and one acts as Client. the follow UUID are be used in
```
#define SERVICE_UUID         "a92ee100-5501-11e4-916c-0800200c9a66"//Keyturner Pairing Service
#define CHARACTERISTIC_UUID  "a92ee101-5501-11e4-916c-0800200c9a66"//Value, General Data Input Output characteristic

```
I can see a simple connection between server and client, and client read the Server CHARACTERISTIC

I don't know how to pair with Encryption....

technyon, 2022-09-27:
Maybe first have a look at this library to understand what's going on and what a client expects the lock to do. First the client wait for a lock in pairing mode by scanning for iBeacons (NukiBle.cpp line 226+):

```
    if (advertisedDevice->haveServiceData()) {
      if (advertisedDevice->getServiceData(pairingServiceUUID) != "") {
        #ifdef DEBUG_NUKI_CONNECT
        log_d("Found nuki in pairing state: %s addr: %s", std::string(advertisedDevice->getName()).c_str(), std::string(advertisedDevice->getAddress()).c_str());
        #endif
        bleAddress = advertisedDevice->getAddress();
        pairingServiceAvailable = true;
      }
    }
```

So when you want the fob to pair, you have to announce this first. When the fob detects an iBeacon, it will start the pairing process ... encryption keys and so on have to be negotiated (NukiBle.cpp, Line 526+). The counterpart to this you can find in the pairing-general-data-io-characteristic.js file I mentioned.

Also check the official documentation:

https://developer.nuki.io/page/nuki-smart-lock-api-2/2

At the very bottom of the document you have examples you can follow step by step ... make sure the numbers match one step at a time. This is actually the difficult part. After that sending commands and so on should be easy.

workhard101, 2022-09-28:
@technyon thank you, I keep trying.

workhard10, 2022-09-28:
Hi, 

I know, it must miss something in my loop()/setup(), I don't know how to go on it, how to lead my code to repair with the NUKI FOB, if I press NUKI FOB 5 Seconds. can anyone help me...

`

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include "HardwareSerial.h"
#include <BLE2902.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "a92ee100-5501-11e4-916c-0800200c9a66"
#define CHARACTERISTIC_UUID "a92ee101-5501-11e4-916c-0800200c9a66"

bool deviceConnected = false;

//Setup callbacks onConnect and onDisconnect
class MyServerCallbacks: public BLEServerCallbacks {
  void onConnect(BLEServer* pServer) {
    deviceConnected = true;
  };
  void onDisconnect(BLEServer* pServer) {
    deviceConnected = false;
  }
};

BLECharacteristic Characteristics("a92ee101-5501-11e4-916c-0800200c9a66", BLECharacteristic::PROPERTY_NOTIFY);
BLEDescriptor     Descriptor(BLEUUID((uint16_t)0x2902));

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

 
  
  BLEDevice::init("ESP32-S3 Server work now");
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE|
                                         BLECharacteristic::PROPERTY_INDICATE|
                                         BLECharacteristic::PROPERTY_NOTIFY
                                       );
  pService->addCharacteristic(&Characteristics);
  Characteristics.setValue("Hello World says Neil");
  Characteristics.addDescriptor(new BLE2902());
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
  
  
}

void loop() {

  Characteristics.notify();
  Serial.print("notify...");
  delay(2000);
}`

technyon, 2022-09-29:
Not sure what's missing here ... have you tried asking in the official Nuki developer forum?

workhard101, 2022-10-03:
@technyon not yet, I have put the question on Nuki developer forum the and wait for the feedback.

I want to renew my question here,. I have annouced and advertised the Server with “Keyturner Pairing Service” with Characteristic(and other all Services and Characteristics in my follow code)

I have already written the follow code, to realize a simple Update for value1, value2, value3 in every Charateristic and it works. I don’t know how can I go on for “Key Pairing” now… I have also read the document “Nuki Smart Lock API V2.1 16.03.2020”, the example about “Command usage” in the document seems what I need, but I don’t know to realize it in C or C++. there is an example " [Smart Lock Simulator](https://developer.nuki.io/t/smart-lock-simulator/70)", it works as NUKI Lock Server and with key pair, but is in Java, it is not easy for me to use it…

@technyon It is possiblel, can you give me a  simple C/C++ example about  how to creat key pair and enable pair mode on server side to pair with a real NUKI FOB, if I press NUKI FOB Buttom 5 second?

I look forward to your feedback. thank you very much in advance!

`#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include "HardwareSerial.h"
#include "Arduino.h"

BLEServer* pServer = NULL;

BLECharacteristic* pKeyturnerPairingGeneralDataInputOutputCharacteristic = NULL;
BLECharacteristic* pKeyturnerGeneralDataInputOutputCharacteristic = NULL;
BLECharacteristic* pKeyturnerUserSpecificDataInputOutputCharacteristic = NULL;

bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value1 = 0;
uint32_t value2 = 0;
uint32_t value3 = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define KEYTURNER_INITIALIZATION_SERVICE_UUID                                  "a92ee000-5501-11e4-916c-0800200c9a66"
#define KEYTURNER_PAIRING_SERVICE_UUID                                         "a92ee100-5501-11e4-916c-0800200c9a66"
#define KEYTURNER_PAIRING_GENERAL_DATA_INPUT_OUTPUT_CHARACTERISTIC_UUID        "a92ee101-5501-11e4-916c-0800200c9a66"
#define KEYTURNER_SERVICE_UUID                                                 "a92ee200-5501-11e4-916c-0800200c9a66"
#define KEYTURNER_GENERAL_DATA_INPUT_OUTPUT_CHARACTERISTIC_UUID                "a92ee201-5501-11e4-916c-0800200c9a66"
#define KEYTURNER_UER_SPECIFIC_DATA_INPUT_OUTPUT_CHARACTERISTIC_UUID           "a92ee202-5501-11e4-916c-0800200c9a66"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("ESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  
  pServer->setCallbacks(new MyServerCallbacks());

    // Create the Keyturner Initialization Service
  BLEService *pKeyturnerInitializationService = pServer->createService(KEYTURNER_INITIALIZATION_SERVICE_UUID);

  // Create the Keyturner Pairing Service
  BLEService *pKeyturnerPairingService = pServer->createService(KEYTURNER_PAIRING_SERVICE_UUID);

  // Create the Keyturner Service
  BLEService *pKeyturnerService = pServer->createService(KEYTURNER_SERVICE_UUID);

  // Create a Keyturner Pairing General Data Input Output Characteristic
[... 55 lines cut ...]
        pKeyturnerPairingGeneralDataInputOutputCharacteristic->notify();
        value1++;

        pKeyturnerGeneralDataInputOutputCharacteristic->setValue((uint8_t*)&value2, 4);
        pKeyturnerGeneralDataInputOutputCharacteristic->notify();
        value2++;

        pKeyturnerUserSpecificDataInputOutputCharacteristic->setValue((uint8_t*)&value3, 4);
        pKeyturnerUserSpecificDataInputOutputCharacteristic->notify();
        value3++;
        delay(3); // bluetooth stack will go into congestion, if too many packets are sent, in 6 hours test i was able to go as low as 3ms
    }
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
}`

technyon, 2022-10-04:
I didn't write the code about key exchange, @I-Connect did. The pointers I can give you:

As already mentioned, study the pairing state machine in the library, and what it expects to happen. A lot of that is in handeReturnMessage. For example, one of the first things is a public key is requested, which is most likely randomly generated. The client sends the command "Command::PublicKey" (line 612, NukiBle.cpp) to request it, you have to check in handleReturnMessage what is supposed to be returned (line 900+) and so on. This unfortunately isnt trivial, and I'd follow the example at the end of the official documentation to see the data matches.

I-Connect, 2022-10-05:
the whole process (both pairing (authorize app) as data exchange including encryption) is described in 
the [Nuki Smart Lock api chapter 9](https://developer.nuki.io/page/nuki-smart-lock-api-2/2/#heading--authorization-data)

workhard10, 2022-10-05:
@technyon @I-Connect 
thanks a lot for the answer!

I have noticed that, in project "[NukiBleEsp32](https://github.com/I-Connect/NukiBleEsp32)", in the pair phase or before the pair is ready the Ble Client receives the data using "notifyCallback", and sends data using in Loop().

I have a question,  in the pair phase or before the pair is ready, how does the server receive the key exchange information, If I press my NUKI FOB 5 seconds, let NUKI FOB be in pair Mode?

in my pinion, use Characteristic in Loop()  to read the key exchange information? 
something like that as follow:

```
#define KEYTURNER_PAIRING_GENERAL_DATA_INPUT_OUTPUT_CHARACTERISTIC_UUID        "a92ee101-5501-11e4-916c-0800200c9a66"

  // Create a Keyturner Pairing General Data Input Output Characteristic
  pKeyturnerPairingGeneralDataInputOutputCharacteristic = pKeyturnerPairingService->createCharacteristic(
                                        KEYTURNER_PAIRING_GENERAL_DATA_INPUT_OUTPUT_CHARACTERISTIC_UUID,
                                                                  BLECharacteristic::PROPERTY_READ   |
                                                                  BLECharacteristic::PROPERTY_WRITE  |
                                                                  BLECharacteristic::PROPERTY_NOTIFY |
                                                                  BLECharacteristic::PROPERTY_INDICATE
}

void loop() {
...
.....
value = pKeyturnerPairingGeneralDataInputOutputCharacteristic->getValue();

pairstatemaschine(value);

}
```

Bascy, 2024-02-12:
Stale issue, closing

closed 2024-02-12 by Bascy