I-Connect/NukiBleEsp32 #23: Opener debug logging contains invalid variables

I-Connect, 2022-09-30:
@technyon on, building with all debug defines on fails, this is due to missing namespaces and there are probably copied items from the lock that do not exist for the opener.

could you please look at this?
(we are also working on the lib so let's coordinate to prevent merge issue)

these are all possible debug defines

DEBUG_SENSE_NUKI
DEBUG_NUKI_COMMAND
DEBUG_NUKI_CONNECT
DEBUG_NUKI_COMMUNICATION
DEBUG_NUKI_HEX_DATA
DEBUG_NUKI_READABLE_DATA

technyon, 2022-09-30:
Sure I'll have a look

technyon, 2022-10-01:
I had a look and could reproduce the issue. I'd like to shift around a few methods ... maybe I better wait until your change is done?

I-Connect, 2022-10-01:
I made some minor changes. Go ahead, let me know when done

technyon, 2022-10-02:
Done ... the changes are actually much smaller than anticipated.

There's something else in there: When calling pairNuki() you can pass the authorization type id so that pairing as an app is possible. A user requested this so the NUKI Hub can be used at the same time as the official Bridge.

I-Connect, 2022-10-02:
merged, but testing still gives errors on building with all debug defines enabled

.pio/libdeps/debug/NukiBleEsp32/src/NukiOpenerUtils.cpp:586:47: error: 'struct NukiOpener::AdvancedConfig' has no member named 'autoLockEnabled'

technyon, 2022-10-02:
I've created another PR. For now I've removed the offending logs for entries in Config and AdvancedConfig. I'll have to add entries the Opener has but the Lock doesn't, I'll do so in another PR ... at least you should be able to compile now.

I-Connect, 2022-10-02:
it compiles now, thx!

closed 2022-10-02 by I-Connect