I-Connect/NukiBleEsp32 #33: New release?

peteh, 2023-02-11:
Hi, there were a bunch of changes, does it make sense to make a new release of the lib to build against? Otherwise projects would break if they build against the main branch.

Bascy, 2023-02-14:
@Jeroen / @technyon Can either of you asses what breaking changes have been made the since 1.0.0 and whether or not we should create a new version here and in the PlatformIO repository

peteh, 2023-02-15:
* Nuki namespace has been renamed to NukiLock
* NukeBle has been renamed to NukiLock
That's the things I had to change to build against the latest master

technyon, 2023-02-15:
As far as I can see my changes to support the opener have been merged since. That means the namespace has to be changed in some places (from Nuki:: to NukiLock:: or NukiOpener::)

Bascy, 2023-02-23:
Created and release a new version v2.0.0 both on github and in the platformio registry

closed 2023-02-23 by Bascy