I-Connect/NukiBleEsp32 #37: Esphome project using this lib sometimes loose connection on commands.

niahane, 2023-03-30:
Hi, I'm using this esphome project which makes use of your library:
https://github.com/uriyacovy/ESPHome_nuki_lock

About 70% of the time the system works perfectly, but sometimes during a command I, but also other people have this error:
[16:45:35][E][nukilock.lock:180]: lockAction failed: 99
[16:45:35][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[16:45:35][D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
[16:45:35][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:36][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:36][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:37][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:37][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:38][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:38][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:39][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:39][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:40][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:40][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:41][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:41][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:42][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:42][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:43][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:43][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[16:45:43][I][nukilock.lock:025]: event notified 0
[16:45:45][I][nukilock.lock:063]: Bat state: 0xc8, Bat crit: 0, Bat perc:100 lock state: 3 14:45:48
[16:45:45][D][lock:055]: 'Nuki Lock': Sending state UNLOCKED
[16:45:45][D][binary_sensor:036]: 'Nuki Connected': Sending state ON
[16:45:45][D][sensor:127]: 'Nuki Battery Level': Sending state 100.00000 % with 0 decimals of accuracy

The developer says it might be worth opening a report to you in hopes of getting more information. Have you ever had reports of this, is there anything I can do to help you?
Thank you.

uriyacovy, 2023-03-31:
Actually the component is using my fork on the dev branch which is only 3 days behind.

I-Connect, 2023-04-03:
ah, check

uriyacovy, 2023-04-07:
I synced both main and dev branch of my fork.
@niahane, can you try again?

stefanschaedeli, 2023-05-10:
The issues seems still to persist according to other issues updates, any ideas ?

peteh, 2023-05-10:
> The issues seems still to persist according to other issues updates, any ideas ?

Are you checking the return value of the command? 

I changed to checking the value and trying to redo the command if it failed for several tries.

zanna-37, 2023-05-13:
I have a similar problem. I'm also using https://github.com/uriyacovy/ESPHome_nuki_lock _but_ reading the code of both repos I suspect that the problem is in this one. Probably in the state machine.

The steps to reproduce the error for me is sending 2 commands near to each other.
The idea is that I unlock the nuki, wait for it to turn, and send a lock command right away.
The result codes are 2 o 3 and sometimes 99. But it never succeeds.

Please, resist the temptation of classifying this as "the ESP is too far" because this only happens when the commands are sent near each other. I don't experience the problem in any other situation, and the lock has been working for months.

I'm open to test, do changes, and compile custom code, but point me in the right direction.

I-Connect, 2023-05-13:
do you wait for the first command to complete before sending the second? Otherwise a semaphore will block the second command as the executeAction method needs to exclusive connection/communication with the lock.

You would see this in the logging if the semaphore fails

zanna-37, 2023-05-13:
I wait for the lock to be still. Then I send the next command.
I understand that there is a semaphore, but in my opinion the second command should be delayed at worse, not fail completely.

zanna-37, 2023-05-13:
At 12:52:34 I send the unlock command
At ~12:52:45 I send the lock command and it fails
At 12:53:12 I send the lock command again and it succeeds

Here are the logs:

[12:52:34][D][lock:054]: 'Nuki Stairs lock': Sending state UNLOCKED
[12:52:53][E][nukilock.lock:180]: lockAction failed: 3
[12:52:53][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[12:52:53][D][lock:054]: 'Nuki Stairs lock': Sending state UNKNOWN
[12:53:03][E][nukilock.lock:075]: requestKeyTurnerState failed: 3
[12:53:03][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[12:53:03][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[12:53:05][I][nukilock.lock:063]: Bat state: 0x94, Bat crit: 0, Bat perc:74 lock state: 3 10:53:30
[12:53:05][D][lock:054]: 'Nuki Stairs lock': Sending state UNLOCKED
[12:53:05][D][binary_sensor:036]: 'Nuki Connected': Sending state ON
[12:53:05][D][sensor:110]: 'Nuki Battery Level': Sending state 74.00000 % with 0 decimals of accuracy
[12:53:12][D][lock:054]: 'Nuki Stairs lock': Sending state LOCKED

zanna-37, 2024-02-12:
The PR uriyacovy/ESPHome_nuki_lock#36 will solve the issue.
The most important thing I did was calling `updateConnectionState()` on each loop.

This function terminates stale Bluetooth connections.
The reason is that, even though [the code states that](https://github.com/I-Connect/NukiBleEsp32/blob/707b327f1b84ab50fbb15b16991a9ea7fc10641d/src/NukiBle.h#L63):
> This method is optional as the lock will also disconnect automatically after ~20 sec.

... the lock is usually not responsive during this time, leading to various errors mentioned in this issue.

Bascy, 2024-02-12:
As this is solved on the side of ESPHome, we can close the issue on this side

zanna-37, 2024-02-12:
If I may, I would suggest to add updateConnectionState() in the example code in the README. What do you think @Bascy?

```
 Nuki::NukiLock nukiLock{deviceName, deviceId};
 BleScanner::Scanner scanner;
 Handler handler;

 void setup() {
   scanner.initialize();
   nukiLock.registerBleScanner(&scanner);
   nukiLock.initialize();
   nukiLock.setEventHandler(&handler);
 }

 void loop() {
   scanner.update();
   delay(10);
   // Terminate stale Bluetooth connections
   updateConnectionState();
 }
```

Bascy, 2024-02-12:
Of course, not a bad idea!
Can you make a pull request?

Bascy, 2024-05-23:
Closing the issue

closed 2024-05-23 by Bascy

zanna-37, 2024-06-01:
> Of course, not a bad idea! Can you make a pull request?

@Bascy, I'm sorry, this slipped under my radar. I've opened a PR #58.