I-Connect/NukiBleEsp32 #41: Cannot save PinCode

ExeErik, 2023-10-29:
I would like to add my Pincode, but I get the following Error:

```
src/main.cpp: In function 'void setup()':
src/main.cpp:186:12: error: 'class NukiLock::NukiLock' has no member named 'savePincode'
   nukiLock.savePincode(1234);
            ^~~~~~~~~~~
```

My Code is the default example:

```

void setup() {
  Serial.begin(115200);
  log_d("Starting NUKI BLE...");
  scanner.initialize();
  nukiLock.registerBleScanner(&scanner);
  nukiLock.initialize();

  if (nukiLock.isPairedWithLock()) {
    log_d("paired");
    nukiLock.setEventHandler(&handler);
    getConfig();
    nukiLock.enableLedFlash(false);
  }

  nukiLock.savePincode("0815");
  // nukiLock.unPairNuki();
}
```

I am grateful for any help

technyon, 2023-10-31:
I think you want to call saveSecurityPincode:

`nukiLock.saveSecurityPincode(1234);`

Where did you find savePincode()?

Also not sure about the pin code for the lock, but for keypad PINs no codes starting with "12" are allowed ... I guess to force people not to use 123456 as the PIN code. Maybe you want to choose some other code for testing.

ExeErik, 2023-11-10:
> I think you want to call saveSecurityPincode:
>
> `nukiLock.saveSecurityPincode(1234);`
>
> Where did you find savePincode()?
>
> Also not sure about the pin code for the lock, but for keypad PINs no codes [...]

Ah, thank you!

I found the "savePincode" on the README.

`Be aware that if you have set a pincode on the lock you will have to store this in the esp using nukiLock.savePincode() otherwise the methods that need a pincode (most methods that write settings) will fail. This only needs to be done once as the pincode will be stored in the preferences.`

closed 2024-02-12 by Bascy