I-Connect/NukiBleEsp32 #49: Crash on Pairing

AzonInc, 2024-04-03:
While pairing a Nuki Lock 4.0 Pro the ESP is crashing right after bringing the Nuki in pairing mode:

Log Output:
```
Sending plain message: 01 00 03 00 27 a7 
Received data: 03 00 18 ca 78 19 39 e2 d0 19 29 02 2b 23 5a 92 8e bc b8 3b 62 8e a8 0b 29 6d 0c 51 2d 9d 3c 2f 4f 42 b5 70 
Remote public key: 18 ca 78 19 39 e2 d0 19 29 02 2b 23 5a 92 8e bc b8 3b 62 8e a8 0b 29 6d 0c 51 2d 9d 3c 2f 4f 42 
Sending plain message: 03 00 ed 3e 97 8b 68 f6 5a de 92 26 10 58 23 0e 13 42 6c 83 58 ae 10 86 dc 46 97 97 11 a6 5a 50 70 14 24 12 
Received data: 04 00 b4 95 aa d9 ae 38 e2 ed c4 96 f0 30 98 6a 69 d3 b7 2f 30 49 c8 4b c5 01 f8 4c 07 79 c3 8d c1 d1 1c c4 
Challenge: b4 95 aa d9 ae 38 e2 ed c4 96 f0 30 98 6a 69 d3 b7 2f 30 49 c8 4b c5 01 f8 4c 07 79 c3 8d c1 d1 
Shared key s: ef a5 78 ee 3b 4e e5 33 54 09 ec b4 ba df 08 55 c9 b7 36 c1 8c 0d ff b9 4d cf c5 23 95 41 8f 77 
Secret key k: 15 0b bd 88 c1 91 b4 a8 b6 76 24 19 d8 26 31 fe fd 41 c7 80 31 00 ec af b2 38 64 27 d1 3d cb c2 
Concatenated data r: ed 3e 97 8b 68 f6 5a de 92 26 10 58 23 0e 13 42 6c 83 58 ae 10 86 dc 46 97 97 11 a6 5a 50 70 14 18 ca 78 19 39 e2 d0 19 29 02 2b 23 5a 92 8e bc b8 3b 62 8e a8 0b 29 6d 0c 51 2d 9d 3c 2f 4f 42 b4 95 aa d9 ae 38 e2 ed c4 96 f0 30 98 6a 69 d3 b7 2f 30 49 c8 4b c5 01 f8 4c 07 79 c3 8d c1 d1 
HMAC 256 result: 81 19 af f3 1e 67 4c 7b 7a 96 e3 51 c3 84 c4 c7 c1 79 8b d0 10 58 b6 65 66 9c 67 c0 88 49 d7 6c 
Sending plain message: 05 00 81 19 af f3 1e 67 4c 7b 7a 96 e3 51 c3 84 c4 c7 c1 79 8b d0 10 58 b6 65 66 9c 67 c0 88 49 d7 6c 6f cc 
E (103311) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (103311) task_wdt:  - loopTask (CPU 1)
E (103311) task_wdt: Tasks currently running:
E (103311) task_wdt: CPU 0: btController
E (103311) task_wdt: CPU 1: IDLE
E (103311) task_wdt: Aborting.

abort() was called at PC 0x400fab4c on core 0

Backtrace:0x40083709:0x3ffbe9bc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

Does this happen because it's a 4th generation lock?

peteh, 2024-04-03:
It looks like you have a task watchdog running but don't reset it in time. 

The lib might take longer than what is currently configured as an allowed time for your watchdog.

peteh, 2024-04-03:
Depends on what watchdog is set on your specific controller. It seems like they can differ depending on what esp you use. 

My guess is still that it got triggered because scanning for bluetooth took a bit longer than the timer allows. 
Using yield instead of delay might also improve the situation.

AzonInc, 2024-04-03:
Thats everything related I could find for ESPHome on ESP32 using Arduino Framework:
```
enableLoopWDT();

  // Disable idle task watchdog on the core we're using (Arduino pins the task to a core)
#if defined(CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0) && CONFIG_ARDUINO_RUNNING_CORE == 0
  disableCore0WDT();
#endif

#if defined(CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1) && CONFIG_ARDUINO_RUNNING_CORE == 1
  disableCore1WDT();
#endif
```

These are the defaults used above:
```
CONFIG_ESP_TASK_WDT=y
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0=n
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1=n
```

AzonInc, 2024-04-03:
Okay I solved it by reconfiguring the watchdog to 15 seconds on Component setup

closed 2024-04-03 by AzonInc