I-Connect/NukiBleEsp32 #65: Support for Smart Lock v4

ssorgatem, 2024-06-27:
It'd be nice to have support for the v4 locks, like nuki_hub does.

It was mentioned here that this library lacks support for v4 locks:

https://github.com/uriyacovy/ESPHome_nuki_lock/issues/49#issuecomment-2195723901

iranl, 2024-06-28:
Nuki Hub also uses this library without modifications for its communication with Nuki devices.

There is no specific code in Nuki Hub or this library needed to make V4 locks work.

So @uriyacovy is mistaken and this library already supports Nuki locks 4 and 4 pro.

If the ESPHome variant does not support V4 locks you should discuss this there as this is no issue with this library.

technyon, 2024-06-28:
Yes, this library does support v4 locks.

closed 2024-06-28 by technyon