I-Connect/NukiBleEsp32 #69: Problem with NukiBle.h

Triskae, 2024-07-05:
Hello everyone,
I'm using this repo to conect to my nuki with esphome [Here](https://github.com/uriyacovy/ESPHome_nuki_lock)
But I have a problem compiling, and I believe the problem come from this dependence :)

Here is the error

```
NukiBleEsp/src/NukiBle.h:332:10: error: 'void Nuki::NukiBle::onDisconnect(NimBLEClient*)' marked 'override', but does not override
```

Can you confirm the problem come from this repo ? :)

Have a nice day

iranl, 2024-07-05:
This is an issue with the esphome implementation of this lib and you should raise your issue there, not here.

Triskae, 2024-07-05:
Okay, thanks for this info ! The issue is also opened there !

closed 2024-07-05 by Triskae