I-Connect/NukiBleEsp32 #76: Example not working

labmaster, 2024-09-25:
When i try the example code (without any Modification) i only get the following error at the serial monitor:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[   287][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[   287][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
[   296][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[   299][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
[   807][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[   807][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
[  1311][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  1312][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
[  1815][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  1816][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
[  2319][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  2320][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
...

Any Hint where to search ?

iranl, 2024-09-25:
What exactly is not working?

These errors are expected when no pairing data is present.

Put the Nuki device in pairing mode and these errors will disappear.

labmaster, 2024-09-25:
I have a Nuki 2.0 and a Nuki 3.0 but none of them "in binding" mode will change this behavior :-(

iranl, 2024-09-25:
You should enable debug logging using the Arduino/ESP-IDF settings in whatever build system you are using to compile and define appropriate debug vars.

Might be better to start from a working implementation such as nuki hub (complex) or esphome_nuki_lock (basic)

labmaster, 2024-09-26:
"nuki hub" is much too complex, would me need to strip down nearly all beside the NUKI plain connect part. "esphome_nuki_lock" is even a big framework (behindESPHome) , that's why i directly choose NukiBleEsp32 as start for a own project.
Just need some basic functions like lock, unlock, unlatch but have to merge it with another framework as soon as i got the nuki function itself running.

I simply download a zip from here, extracted it, opened it in VSCode/PlatfomIO, build the example and uploaded it (to different ESP32 Wroom Boards with 4MB Flash) with the result i showed above.

As quickcheck i downloaded and build and uploaded "nuki hub" only to check if the locks and the ESP32 Hardware is working as expected. 
It runs out of the box !!!

I never used debugging on a ESP32 (beside Serial) so no clue at the moment to to debug this problem. Not sure how to continue now.

labmaster, 2024-09-26:
Here are serial two output in debug build:  (cut from long serial output)

1. snip ###########################################
Nuki V2 was in bind mode:
...
I NimBLEScan: New advertiser: ca:9c:54:9d:a8:5a
D NimBLEScan: erase device: ca:9c:54:9d:a8:5a
I NimBLEScan: New advertiser: 70:2a:d5:ff:9d:91
D NimBLEScan: erase device: 70:2a:d5:ff:9d:91
I NimBLEScan: New advertiser: 2f:d3:87:77:9f:aa
D NimBLEScan: erase device: 2f:d3:87:77:9f:aa
[457396][E][Preferences.cpp:526] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[457407][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
I NimBLEScan: New advertiser: 68:52:c9:e0:fc:53
D NimBLEScan: erase device: 68:52:c9:e0:fc:53
[457423][D][NukiBle.cpp:136] pairNuki(): No nuki in pairing mode found
[457429][D][NukiBle.cpp:141] pairNuki(): pairing result 0
I NimBLEScan: New advertiser: ca:9c:54:9d:a8:5a
D NimBLEScan: erase device: ca:9c:54:9d:a8:5a
I NimBLEScan: New advertiser: cb:17:a6:a6:85:83
D NimBLEScan: erase device: cb:17:a6:a6:85:83
I NimBLEScan: New advertiser: 2f:d3:87:77:9f:aa
D NimBLEScan: erase device: 2f:d3:87:77:9f:aa
I NimBLEScan: New advertiser: 4b:cd:31:99:9a:24
D NimBLEScan: erase device: 4b:cd:31:99:9a:24
I NimBLEScan: New advertiser: 54:d2:72:f7:4b:e4
I NimBLEAdvertisedDevice: No service data found    **************************
D NimBLEScan: erase device: 54:d2:72:f7:4b:e4
I NimBLEScan: New advertiser: ca:9c:54:9d:a8:5a
D NimBLEScan: erase device: ca:9c:54:9d:a8:5a
I NimBLEScan: New advertiser: 2f:d3:87:77:9f:aa
D NimBLEScan: erase device: 2f:d3:87:77:9f:aa
I NimBLEScan: New advertiser: cb:17:a6:a6:85:83
D NimBLEScan: erase device: cb:17:a6:a6:85:83
I NimBLEScan: New advertiser: ca:9c:54:9d:a8:5a
D NimBLEScan: erase device: ca:9c:54:9d:a8:5a
I NimBLEScan: New advertiser: 68:52:c9:e0:fc:53
D NimBLEScan: erase device: 68:52:c9:e0:fc:53
I NimBLEScan: New advertiser: 2f:d3:87:77:9f:aa
D NimBLEScan: erase device: 2f:d3:87:77:9f:aa
I NimBLEScan: New advertiser: 54:d2:72:f7:4b:e4
I NimBLEAdvertisedDevice: No service data found    **************************
D NimBLEScan: erase device: 54:d2:72:f7:4b:e4
I NimBLEScan: New advertiser: 4b:cd:31:99:9a:24
D NimBLEScan: erase device: 4b:cd:31:99:9a:24
I NimBLEScan: New advertiser: 70:2a:d5:ff:9d:91
D NimBLEScan: erase device: 70:2a:d5:ff:9d:91
I NimBLEScan: New advertiser: 59:cd:10:72:3c:ae
D NimBLEScan: erase device: 59:cd:10:72:3c:ae
I NimBLEScan: New advertiser: d4:9d:c0:bd:28:57
D NimBLEScan: erase device: d4:9d:c0:bd:28:57
I NimBLEScan: New advertiser: 2f:d3:87:77:9f:aa
D NimBLEScan: erase device: 2f:d3:87:77:9f:aa
I NimBLEScan: New advertiser: cb:17:a6:a6:85:83
D NimBLEScan: erase device: cb:17:a6:a6:85:83
I NimBLEScan: New advertiser: ca:9c:54:9d:a8:5a
D NimBLEScan: erase device: ca:9c:54:9d:a8:5a
I NimBLEScan: New advertiser: ee:a0:32:0f:fe:e5
D NimBLEScan: erase device: ee:a0:32:0f:fe:e5
I NimBLEScan: New advertiser: 2f:d3:87:77:9f:aa
D NimBLEScan: erase device: 2f:d3:87:77:9f:aa
[457935][E][Preferences.cpp:526] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
I NimBLEScan: New advertiser: 54:d2:72:f7:4b:e4
I NimBLEAdvertisedDevice: No service data found    **************************
D NimBLEScan: erase device: 54:d2:72:f7:4b:e4
I NimBLEScan: New advertiser: d4:9d:c0:bd:28:57
D NimBLEScan: erase device: d4:9d:c0:bd:28:57
I NimBLEScan: New advertiser: cb:17:a6:a6:85:83
D NimBLEScan: erase device: cb:17:a6:a6:85:83
I NimBLEScan: New advertiser: 70:2a:d5:ff:9d:91
[... 90 lines cut ...]
D NimBLEScan: erase device: 54:d2:72:f7:4b:e4
I NimBLEScan: New advertiser: 68:52:c9:e0:fc:53
D NimBLEScan: erase device: 68:52:c9:e0:fc:53
I NimBLEScan: New advertiser: 2f:d3:87:77:9f:aa
D NimBLEScan: erase device: 2f:d3:87:77:9f:aa
I NimBLEScan: New advertiser: 59:cd:10:72:3c:ae
D NimBLEScan: erase device: 59:cd:10:72:3c:ae
I NimBLEScan: New advertiser: ca:9c:54:9d:a8:5a
D NimBLEScan: erase device: ca:9c:54:9d:a8:5a
I NimBLEScan: New advertiser: 4b:cd:31:99:9a:24
D NimBLEScan: erase device: 4b:cd:31:99:9a:24
I NimBLEScan: New advertiser: 2f:d3:87:77:9f:aa
D NimBLEScan: erase device: 2f:d3:87:77:9f:aa
I NimBLEScan: New advertiser: cb:17:a6:a6:85:83
D NimBLEScan: erase device: cb:17:a6:a6:85:83
...

So if the lock is in binding mode then it looks like this: (and 54:d2:72:f7:4b:e4 appears more often than if not in binding mode)
I NimBLEScan: New advertiser: 54:d2:72:f7:4b:e4
I NimBLEAdvertisedDevice: No service data found
D NimBLEScan: erase device: 54:d2:72:f7:4b:e4

If the lock is not in binding mode then it looks like that
I NimBLEScan: New advertiser: 54:d2:72:f7:4b:e4
D NimBLEScan: erase device: 54:d2:72:f7:4b:e4

labmaster, 2024-09-26:
I now installed a fresh VSCode with fresh installed PlatformIO .
Again downloaded "https://github.com/I-Connect/NukiBleEsp32/archive/refs/heads/main.zip" and built the example in debug mode. 
Result is the same 
This was my last hope that i had done something wrong  :-(

labmaster, 2024-09-27:
Below is a serial output from the powerup, a Nuki V3 was in bind mode, so the typical 
"NimBLEAdvertisedDevice: No service data found" was there. 
I marked it with "******************"

I tracked down the message "No service data found" in sourcecode "NimBLEAdvertisedDevice.cpp"
there in ~line 345 at " if(bytes == uuidBytes && NimBLEUUID(field->value, bytes, false) == uuid) {"
the comparison between "NimBLEUUID(field->value, bytes, false)" and "uuid" always fails.
Where bytes and uuidBytes are always  "24"

I still think that something is wrong or missing in the example code.
Please Help

##########################################
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
[     8][D][esp32-hal-cpu.c:264] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 100
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   278744 B ( 272.2 KB)
  Free Bytes        :   247528 B ( 241.7 KB)
  Allocated Bytes   :    25736 B (  25.1 KB)
  Minimum Free Bytes:   242492 B ( 236.8 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : DIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
[... 46 lines cut ...]
I NimBLEScan: New advertiser: 70:2a:d5:ff:9d:91
D NimBLEScan: erase device: 70:2a:d5:ff:9d:91
I NimBLEScan: New advertiser: cb:17:a6:a6:85:83
D NimBLEScan: erase device: cb:17:a6:a6:85:83
I NimBLEScan: New advertiser: 7a:37:7c:96:60:8e
D NimBLEScan: erase device: 7a:37:7c:96:60:8e
I NimBLEScan: New advertiser: 58:1e:85:76:76:03
D NimBLEScan: erase device: 58:1e:85:76:76:03
I NimBLEScan: New advertiser: ca:9c:54:9d:a8:5a
D NimBLEScan: erase device: ca:9c:54:9d:a8:5a
I NimBLEScan: New advertiser: 54:d2:72:4e:3c:dd
I NimBLEAdvertisedDevice: No service data found  ******************
D NimBLEScan: erase device: 54:d2:72:4e:3c:dd
I NimBLEScan: New advertiser: 53:1e:0e:1f:01:f2
D NimBLEScan: erase device: 53:1e:0e:1f:01:f2
I NimBLEScan: New advertiser: cb:17:a6:a6:85:83
D NimBLEScan: erase device: cb:17:a6:a6:85:83
I NimBLEScan: New advertiser: 6b:ce:69:3e:48:6c
D NimBLEScan: erase device: 6b:ce:69:3e:48:6c
I NimBLEScan: New advertiser: 70:2a:d5:ff:9d:91
D NimBLEScan: erase device: 70:2a:d5:ff:9d:91
I NimBLEScan: New advertiser: ca:9c:54:9d:a8:5a
D NimBLEScan: erase device: ca:9c:54:9d:a8:5a
I NimBLEScan: New advertiser: 61:27:2d:e6:22:f5
D NimBLEScan: erase device: 61:27:2d:e6:22:f5

labmaster, 2024-09-27:
Finally found this older issue ( https://github.com/I-Connect/NukiBleEsp32/issues/38 )

That gave the hint to a specific branch "pair-and-unpair-refactor"  ( https://github.com/I-Connect/NukiBleEsp32/tree/pair-and-unpair-refactor )

I tried it and this one immediatly worked out of the box !!!

labmaster, 2024-09-27:
NEWS !!! 
by comparing  
https://github.com/I-Connect/NukiBleEsp32/tree/pair-and-unpair-refactor 
with
https://github.com/I-Connect/NukiBleEsp32/archive/refs/heads/main.zip

I tracked down the binding problem to only a couple of lines where the UUIDs are defined.
Its all about NukiLock.cpp at the very beginning:

```
NukiLock::NukiLock(const std::string& deviceName, const uint32_t deviceId)
  : NukiBle(deviceName,
            deviceId,
            keyturnerPairingServiceUUID,
            keyturnerServiceUUID,
            keyturnerGdioUUID,
            keyturnerUserDataUUID,
            deviceName) {
    errorCode = (uint8_t)ErrorCode::ERROR_UNKNOWN;
}
```

If i change this to:

```
NukiLock::NukiLock(const std::string& deviceName, const uint32_t deviceId)
  : NukiBle(deviceName,
            deviceId,
            NimBLEUUID("a92ee100-5501-11e4-916c-0800200c9a66"),
            NimBLEUUID("a92ee200-5501-11e4-916c-0800200c9a66"),
            NimBLEUUID("a92ee101-5501-11e4-916c-0800200c9a66"),
            NimBLEUUID("a92ee202-5501-11e4-916c-0800200c9a66"),
            deviceName) {
    errorCode = (uint8_t)ErrorCode::ERROR_UNKNOWN;
}
```

then binding works as it should  !!!

technyon, 2024-09-29:
I have no explanation why this makes a difference, both should create identical NimBLEUUID objects. Could you add some debug output inside NukiBle.cpp for testing, to see if there's a difference. The NimBLEUUID class has a toString() method you can use. Both ways should have identical output.

labmaster, 2024-09-29:
No, as far i could find, both create different NimBLEUUID objects.

I tried to output the UUID in a readable way, but even with .toString i only had partly success, so i dumped the object byte by byte as hex char now.

labmaster, 2024-09-29:
For a test i changed the following:
in Nukilock.cpp
`
NukiLock::NukiLock(const std::string& deviceName, const uint32_t deviceId, const NimBLEUUID PairingServiceUUID, const NimBLEUUID ServiceUUID, const NimBLEUUID GdioUUID, const NimBLEUUID UserDataUUID)
  : NukiBle(deviceName,
            deviceId,
            PairingServiceUUID,
            ServiceUUID,
            GdioUUID,
            UserDataUUID,
            deviceName) {
    errorCode = (uint8_t)ErrorCode::ERROR_UNKNOWN;
}
`
in Nukilock.h
`NukiLock(const std::string& deviceName, const uint32_t deviceId, const NimBLEUUID PairingServiceUUID, const NimBLEUUID ServiceUUID, const NimBLEUUID GdioUUID, const NimBLEUUID UserDataUUID);
`
and in NukiSmartlockTest.h  (-> main.cpp)
`NukiLock::NukiLock nukiLock(deviceName, deviceId, NukiLock::keyturnerPairingServiceUUID, NukiLock::keyturnerServiceUUID, NukiLock::keyturnerGdioUUID, NukiLock::keyturnerUserDataUUID);
`
This will even work then.

For me it looks like as if at the time when the child object NukiLock" is created from "NukiBle" ( see "NukiLock.cpp NukiLock.h")  the NimBLEUUID object was not already evaluated.

technyon, 2024-10-01:
Strange, I can't really make sense of it, I'll have to think about it.

dersvenni, 2025-11-13:
Hello,
i observed similar things and i found out, that the problem was for me the name of the device. 
I have two locks which i had test with the same esp. If one lock was connected and i try the other one, i was not able to pair it. I observed the same in the logs as above, that no service data was found. If i erase the flash then it was working. But it also was working if i change the device name. Maybe that hint helps somehow.

Sven