I-Connect/NukiBleEsp32 #86: IDF support

AzonInc, 2025-02-13:
Hi,

I created a fork and converted it to an idf only component.
Do you think there is a way of merging both together to be an idf component and arduino library in one codebase?

its here: https://github.com/AzonInc/NukiBleEsp32

iranl, 2025-02-15:
Of course it is possible to have support for ESP-IDF and Arduino (and Arduino as an IDF-component).

Probably moving the logging functions to a separate class with guards based on defines that Arduino sets and ESP-IDF doesn't. Your current solution imho is not the right way to implement this in a non breaking way for both Arduino and IDF.

Also copying Arduino libraries (Preferences) to this lib kinda defeats the purpose imho while also reducing maintainability. There is also limited benefit when comparing your solution to using Arduino as an IDF component (while using as little Arduino components as necessary).

I see you are already looking into solutions for using Arduino as an IDF component in ESPHome. It's a shame this doesn't seem to have a lot of traction atm.

I don't see us accepting any changes that include Arduino libraries both out of principle and as you are currently not compliant with the license of the Preference library

AzonInc, 2025-02-15:
Yea I'm aware that this is not the right solution. It was just a test to be able to have ESPHome running without any arduino library code which works perfectly fine for now and also reduced the flash and memory use quite a bit.

The Preferences lib shouldnt be in there but is temporarily there until i find another solution.

technyon, 2025-02-16:
Hi,

We could for now mention your fork in the readme so people are aware of it. That would mean though you'd need to support this fork long term, until we find a solution to have both in one repository. If you're fine with that, we can add a link.

AzonInc, 2025-02-16:
I'm actively using it in my project so probably also add the new changes whenever they become available.
Sounds good to me, I will try to make the fork more compatible with the base arduino one as much as possible by using some flags and restoring existing ones.

technyon, 2025-02-19:
ok, let me know when your fork is ready. Also let's update the readme in that fork make clear what it is and maybe add some basic documentation.